frangoteam/FUXA #675: Add Internationalization of zh-cn

feast107, 2023-01-29:
Nice to meet you.
It seems Chinese internationalization resource has been un-updated for long.
What if I put some translation in which will be match with the latest en.json. Some unproper translation shall be fixed at the same time

feast107, 2023-01-29:
By the way, Im working on connecting to SignalR Server as a new type of device, which is microsoft's real-time application and works like Websocket. Is there any possibility to add it into the main? The workflow can be seen in my fork.

unocelli, 2023-01-29:
Hi,
of course you can update and complete the zh-cn translation, you are welcome.
on the subject of signalR we find it interesting. in which context you want to use it?

feast107, 2023-01-30:
In ```Device Setting``` that can add new device whose type is ```signalR``` ,just use it to watch real-time value like mqtt's 'pub/sub' but working on another transportation protocol and a certain server.

unocelli, 2023-02-05:
do you plan to define your own tag object?

closed 2023-03-03 by feast107