frangoteam/FUXA #679: background color of views

mariorl, 2023-01-31:
Hi, I noticed that the background color chossen in LAYOUT SETINGS/GENERAL/START VIEW force all other views with the same background color.

unocelli, 2023-02-05:
Hi, the background will be dynamically synchronised with the background of the current view

mariorl, 2023-03-15:
I just noticed, if the user goes to an HMI page from a Main Index page, the  background is taken from the background color of page declared in the general layout / start view field. 
But if you access an HMI page using a "named view"  such "http://127.0.0.1:1881/view?name=NAME OF THE PAGE", then the background color is OK.

unocelli, 2023-03-21:
Hi, the background color of Home inherits the color from the current View showed