frangoteam/FUXA #680: Operator or Viewer only access

ViolentPillow, 2023-02-01:
I may be a complete fool, but I am failing to understand how an operator level person can view this. I only have administrator access all the time, but I want this revoked for operators. So, how do I access the server as an operator only? I setup an operator user, but I don't see this available. Can you remotely view this from another machine? P.S. this is an awesome HMI quick development environment I am using with Click PLC very easily.

Obaq-web, 2023-02-02:
The operator's computer needs to be in the same network as the server where fuxa is running on.

To connect from, lets say a mobile 5G-device, you need a VPN to the network where the server is running.

ViolentPillow, 2023-02-04:
Forgive me, yes, my network was set to public and not private. My firewall was quietly swatting me. So, sorry!

unocelli, 2023-03-19:
I am going to close this as resolved, let us know if you have any issues.

closed 2023-03-19 by unocelli