frangoteam/FUXA #681: Why is my menu missing some options

pengju97, 2023-02-02:
[image]
https://frangoteam.github.io/
domo
[image]
I was deployed by docker，How to solve this p
blem？？

unocelli, 2023-02-05:
Hi, which version are you using?

pengju97, 2023-02-06:
[image]

pengju97, 2023-02-06:
[image]

pengju97, 2023-02-07:
> Hi, which version are you using?

[image]

unocelli, 2023-02-07:
You need to install the latest version

closed 2023-02-07 by unocelli