frangoteam/FUXA #682: Modbus bug when reading two consecutive registers

IonuDragomir1, 2023-02-03:
Hi,
I'm facing a bug when trying to read the value of two registers as a float.
I read through Modbus RTU from a Schneider power meter using a Schneider PLC. I connect Fuxa to this PLC using Modbus TCP. I have to read a voltage parameter on two registers. What I noticed is that when read each register, I get correctly the values. The problem appears when I want to read both registers as a float value. So when I add the third tag to read both registers from the first register's offset, I get its value instead of the value of the two registers. In Fuxa I've tried to replace the Float32 data type with another double word data type, but the result is the same.  
For example:
WordReg1 (address: 400501):  20895 
WordReg2 (address: 400502): 17258
FloatValue (address: 400501): 20895 
So as you can in above example, FloatValue it's not containing both registers (WordReg1 and WordReg2) and convert them in float.
I've run the same test in WinCC and there I get the correct value in the Float32 data type tag. Please see the image attached with the working test I made in WinCC.
Please also see the attached video where you can see the values shown in PLC and the ones shown in Fuxa. I use Fuxa version: 1.1.11/1064; Angular version: 11.2; nodjs:14.18.
[image]

https://user-images.githubusercontent.com/58516894/216641792-ee973169-04f9-4b50-9ff2-3e6ad4da781c.mp4

Thank you very much in advance!

unocelli, 2023-02-05:
Hi, currently configure more tags at the same address, particularly if they are of two different types, is not supported.
you have to choose:
- WordReg1 (address: 400501): 20895
- WordReg2 (address: 400502): 17258
or
- FloatValue (address: 400501): 20895

IonuDragomir1, 2023-02-06:
Hi,
I think you misunderstood me. WordReg1 and WordReg2 are the same type
FloatValue should contain both registers (WordReg1 and WordReg2). So basically I want to read with one tag (FloatValue) the two registers. It's not OK that FloatVaule which is a 32 bit represented Float has the same value as WordReg1 which is 16 bit represented word. When I read FloatValue, the two registers should be concatenated and converted to float as you can see that happens in WinCC.  If instead of FloatValue, I'd have a DintValue, the two registers should be concatenated and converted to dint.

IonuDragomir1, 2023-02-10:
Beside the scenario I connect Fuxa to PLC and the PLC is connected to whatever other device, there is the possibility of connection Fuxa directly to power meter through Modbus RTU.
The power meter's parameter  values can be read using 32 bits tags (current, voltage, etc) or 64 bits tags (energy, etc). Please see the below screen shots from power meter's documentation. 
The 2 registers (words) parameters:
[image]

The 4 registers (words) parameters:
[image]

Coming back to my provided example, this is why when I read FloatValue, the two registers should be concatenated and converted to float. The same thing applies to 64 bits tags in case I want to read energies.

robbudge, 2023-02-11:
Try and shift your request register. 
I use floats all the time but often run into reverse word issues. 
when you set a data type of float the request is always 2 registers. 
the question becomes is it target address & address +1 or target address & target -1. 
Also what register is the high register in the 32bit data block. 

I will quite often load a float with a generic number 12.34567 into the controller then convert it to an IEC 32 float in raw hexadecimal. 
Then read the word registers in raw format to confirm the byte / word order. 

 I’m not sure is Fuxa or Schiender have alternative ‘Float’ options to adjust the word / byte order.

robsori, 2023-02-13:
I agree that Fuxa should be able to read consecutively two or four registers for the case described by IonuDragomir1: two consecutive registers for electrical quantities (currents, voltages) and four consecutive registers for active and reactive energies from the measuring stations. The values read for the electrical quantities described must be assigned to some Fuxa tags.
I have also encountered this case in SCADA applications where the electrical quantities from the power plants for measuring the consumed energy must be displayed.

unocelli, 2023-02-21:
Hi, 
the feature still needs to be implemented, in the meantime I think you can solve the issue with the scripts.

henjoe, 2024-01-23:
Hi @unocelli , 
Is there an update to this?

unocelli, 2024-10-27:
Hi @henjoe, some changes have been made on modbus, please confirm if the problem still occurs

closed 2025-12-26 by unocelli