frangoteam/FUXA #683: server install error

liuweibing7, 2023-02-04:
when I run `npm install` in server dictionary, an error occurred：
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! sqlite3@5.0.11 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the sqlite3@5.0.11 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\A\AppData\Roaming\npm-cache\_logs\2023-02-04T10_57_28_407Z-debug.log

env:
node version: v14.2.0
npm version: 6.14.4

I am new to learn Fuxa and I don't know how to fix it. What's the problem?

Rafael-OliveiraBR, 2023-02-06:
> when I run `npm install` in server dictionary, an error occurred： npm ERR! code ELIFECYCLE npm ERR! errno 1 npm ERR! sqlite3@5.0.11 install: `node-pre-gyp install --fallback-to-build` npm ERR! Exit [...]

Your npm and nodejs version look a bit older.
Try update npm and node.

----
npm install -g n
n latest
----

Or you can download directly from https://nodejs.org/en/

liuweibing7, 2023-02-07:
Yes you are right. I installed newer version of node, it worked. Thanks.

closed 2023-02-07 by liuweibing7

CHONGE, 2024-01-10:
use yarn;success