frangoteam/FUXA #688: Alarms Inactive

robbudge, 2023-02-11:
My Project is running Version 1.1.10-1053.
I have alarms Configured and even tried to follow the example. No Alarms are firing even with them showing enabled.
[image]

[image]
[image]
No Alarms 
[image]
[image]

Am I missing something with enabling the Alarm engine ???

unocelli, 2023-02-11:
Hi,
you have to set a Time Delay, how long has to be in range to trigger the alarm (0 = disabled)

robbudge, 2023-02-11:
I did also run alarms with a delay of 1 second.

Best regards,
Robert Budgen
Business Development Manager: Process Control Solutions

Bürkert Fluid Control Systems
Burkert Contromatic Inc.
5002 South Service Road<x-apple-data-detectors://4/1>
Burlington ON  L7L 5Y7<x-apple-data-detectors://4/1>
Cell:  (905) 630-0694<tel:(905)%20630-0694>
Phone: (905) 632-3033<tel:(905)%20632-3033>
Fax:     (905) 632-3833<tel:(905)%20632-3833>
***@***.******@***.***>
www.burkert.ca<http://www.burkert.ca/>

Sent from IPhone

unocelli, 2023-02-14:
I agree, Time Delay is not very clear.
I am not sure if I understand if the alarms now work?
if you prefer, you can send us the project (json export) so that we can analyse the problem.

robbudge, 2023-02-14:
Sure, this  is just a scratch project. 
Back working on some Rockwell stuff. its so Basic it hasn't changed in over 20yrs. Once I get over this stupid hurdle in Logix Designer I will pull the OpenPLC project and have another look for errors.

[fuxa-project-2.json.zip](https://github.com/frangoteam/FUXA/files/10737413/fuxa-project-2.json.zip)

if you guys ever need access to industrial hardware DM me.
My Test lab has a VPN and the following hardware.
IFM Asi Master- Modbus TCP.
Compact Logix EthernetIP.
2  x OpenPLC RPI's
Wago 750-82xx PLC Running Codesys.
Burkert ME63 CanOpen to Industrial Ethernet Gateway with devices can run Profinet / EthernetIP/ Modbus / OPCua
Graphite TSDB with Grafana DashBoard.
just to mention a bit.

unocelli, 2023-02-17:
Hi, 
I have fixed the issue with the timer delay (it is now optional) and the alarm trigger was also not very optimal. 
Thank you for the very interesting proposal, please take contact with us directly 4frango@gmail.com

unocelli, 2023-02-25:
Hi,
is the problem with the alarms solved?

unocelli, 2023-03-19:
I am going to close this as resolved, let us know if you have any issues.

closed 2023-03-19 by unocelli