frangoteam/FUXA #689: Lots of data

576204520, 2023-02-13:
Hello, how many devices can be supported and how many tags of data can be obtained in real time, because I see that it is a sqlite database, and the performance will be limited

unocelli, 2023-02-13:
Hi, depend on platform, settings and you can use influxdb.

576204520, 2023-02-14:
I know that real-time data can be put into the influxDB database, but the point configuration data is in sqlite, I configure a device with more than 10,000 points, there will be broken links and page stuttering, is there any way

unocelli, 2023-02-14:
I do not think that the project data can be a performance problem. there are certainly improvements that can be made to the data transmitted to the frontend.