frangoteam/FUXA #690: Values not stored, how to debug

ProfDrOetker, 2023-02-13:
I have a fuxa server with a working S7 connection, so tag values are shown in the tag list, but the values are not stored for historic charts (live charts are working). I did enable the registration of the tags.

In the server console no errors or warnings are shown, so I am stuck and do not know where to look to find the problem.

Setup:
Ubuntu 20.04.5
npm 6.14.4
Fuxa 1.1.11-1075
Firefox 109.0.1

ps: what a great system this is, if the history is working, I doubt if I will ever use something else

unocelli, 2023-02-13:
Hi, we have to check...is important to enable the tags registration
[image]

ProfDrOetker, 2023-02-14:
registration is enabled, and I did a reinstall of node and npm, and now values FUXA Server values are recorded (registrated) in history, but all other PLC tags/variables are not:
[image]

unocelli, 2023-02-14:
is it possible that you have configured another tag? if I see from the screenshots you have a chart with a tag IW134 and in device a IW130. otherwise you can send us the project, exported in json format.

ProfDrOetker, 2023-02-14:
I have created many tags and tested InfluxDB, and there the same issue arises: all tags are recorded in history but not the S7 connected device tags. I will email the project

ProfDrOetker, 2023-02-15:
Partly solved: FUXA/Snap7 seems to need 'absolute' addresses to enable recording of values. E.g `DB3.DW2` works perfect for displaying and recording, but `IW134` only shows actual values, and no recording takes place. I think this classifies as a bug.

unocelli, 2023-02-15:
Yes, in fact it is a bug, we will fix it, Thanks for the report.
maybe you can help us, our plc simulator support only DB area variables and therefore we cannot test it with PE/PA/MK area. do you know of any freeware software that can simulate the s7 protocol and PE/PA/MK variables (like IW134)?

ProfDrOetker, 2023-02-15:
You're welcome! I am not aware of any simulation software for s7 hardware.

mariorl, 2023-03-08:
> Yes, in fact it is a bug, we will fix it, Thanks for the report. maybe you can help us, our plc simulator support only DB area variables and therefore we cannot test it with PE/PA/MK area. do you know of any freeware software that can simulate the s7 protocol and PE/PA/MK variables (like IW134)?

Hi, I got the same problem, in fact this is the reason of not able to finish my Fuxa project. You can easily emulate actual S7 comms in the whole S7 memory map (Merkers, input, output, DB...) for free, using PLCSIM + Network to PLCsim packages.  If somebody is interested, I have a small guide in order to setup these packages saving some time.

unocelli, 2023-03-09:
Hi, I'm not sure if you are talking about the same issue, something has been fixed, with which version have you tested?
About the simulator, yea we are interested to a guide.

mariorl, 2023-03-15:
Yeah, sorry by my side, I installed the last version and now datalloging - graphs work OK.

mariorl, 2023-03-15:
> Hi, I'm not sure if you are talking about the same issue, something has been fixed, with which version have you tested? About the simulator, yea we are interested to a guide.

NetToPlcSim is a Window sw capable to interconnect S7-PLCsim with ethernet client devices, as HMIs or Scadas.
When you set up NetoToPlcSim is recommended to follow these following steps. Please note, the order of them matters.

1. Start step7, and open your project. Your project must include an ethernet conectivity node such as CP343-1, native port etc.
2. Start NetoToPlcSim as Administrator. (It needs it in order to gain access to port 102, let it to do). 
3. On Step 7, make click on the PLCsim lauch simulation icon.
4. Once PLCsim is up, download your project. Download HW config also, this will leave PLCsim accesible for NetoToPlcSim in the CP343-1 IP address.
5. In NetoToPlcSim user interface you need now to create a connection. NetoToPlcSim needs to know the CP343-1 IP, click in ADD button.
6. A dialog appears in order to set up a connection. In the "network IP address", by click the "..." button, NettoPlcSim will suggest PC's detected network cards.
7. Choose any network card, physical or emulated.
8. Click on PLCsim IP address "..." button. If everithing is OK, the CP IP address will appear there. 
9. Please note, for S7300 rack 0 Slot 2, for S71200, rack 0 slot 1, etc.
10. Click OK and then "Start server".
11. Make sure your HMI project S7 connection is pointing to the IP address declered in "network IP address" field declared during connection setup.
12. There is a monitoring function during HMI-PLCsim communications, S7 packets are shown, very useful for comms debug. (Right-click mouse over connection)

Make sure "PLCsim IP address" and "network IP address" are not in the same range.
Of course PLCsim could be inside a VM, and FUXA in the host. In this case NetoToPlcSim will not detect PLCsim, you should enter the VM IP manually.
It is possible to connect an HMI and debug with Step7 at the same time.
Enjoy

unocelli, 2023-03-15:
I will try it, Thanks a lot

unocelli, 2023-03-19:
I am going to close this as resolved, let us know if you have any issues.

closed 2023-03-19 by unocelli