frangoteam/FUXA #692: String S7

DIMIIIII, 2023-02-15:
Hi Fuxa team,

I found a Data Type that is not available on Fuxa, on a Snap7 connection, it is impossible to bind a Type String variable, Is it in your projects to add it?

Thank you.

unocelli, 2023-02-21:
Hi, 
this request is not new, there is a CHAR type that we will try, it is 1 byte ascii format.

DIMIIIII, 2023-02-22:
Can you tell me when it will be available?
Thank you

unocelli, 2023-02-28:
to confirm, we would be interested to know how you configure a string address with S7?

DIMIIIII, 2023-03-01:
I can't really understand, normally, it has the bytes or bytes to join, it seems to me that I had found on snap7 the way to read strings in s7 MultiVar, but impossible to get my hands on it.

audimex, 2023-08-23:
> to confirm, we would be interested to know how you configure a string address with S7?

this is how I normally do:
STRING[20],DB20.DBX4.0

[20] means 20 bytes (char), in S7 a string is just a char array + 2 additional bytes:

B0: total length of the string (in this example: 20)
B1: used length on the string (number of characters actually used, "hello" -> 5)
B2: first char of the string
B3: second char of the string
...
B21: 20th char of the string

unocelli, 2023-08-24:
Hi, data are missing, to which db should they be read and from which db index?

audimex, 2023-08-24:
This is a syntax example I use:
STRING[20],DB20.DBX4.0

breakdown:
STRING[20] = data type (STRING) + string max. length([20])
DB20 = points to data block 20
DBX4.0 = points to byte 4 and bit 0 of data block 20

So first character of the string starts at DB 20 byte 4 and spans up to 22 bytes (2 control bytes + 20 bytes)