frangoteam/FUXA #693: i want to add my own svg menu

patson12, 2023-02-16:
hello ,
i want to add my own svgs in menu .. or i want _images folder from client side right now i cant get this folder i get that error in error.jpg file
![error](https://user-images.github
[image]
usercontent.com/96180028/219283630-e62f297c-1a93-4260-85ca-2e524fd31a6f.JPG)

i want svgs like this from client side. which is only getting rightnow from server side.
[image]

unocelli, 2023-02-16:
Hi, you are in debug mode (http://localhost:4200), you have to use production port http://localhost:1881

patson12, 2023-02-17:
what if i want to use http://localhost:4200/ ?? can ypu plz help me?

unocelli, 2023-02-17:
url with port 4200 is used from angular to serve the frontend application in development and your resource (images) are available in fuxa web server (nodejs/express), url displayed in the console when starting the server.

patson12, 2023-02-21:
ok done.. thanx

closed 2023-02-21 by patson12