frangoteam/FUXA #694: polling an overload in s7

leeallen1988, 2023-02-16:
Hello,
I met a same issue.
When I'm trying to add more than 20 tags in a device which connects to a 1500 PLC, I'm getting " __readVars error! TypeError: Array exceeds max variables (20) that can be transferred with ReadMultiVars()_" error message (please see below image). For this connection I use SiemensS7 protocol, Fuxa version is 1.1.10 (I have tried also the version 1.1.9, but the error persists) . This error occurs when I add 21st tag and no value is displayed anymore until only 20 tags remain.
the lastest version 1.1.12 has the same situation. please help ,Thanks in advance.
.

ProfDrOetker, 2023-02-16:
I tested with 21 tags on a S7-1200 PLC and this worked fine if the tags are all DB addressed, like `DB8.DBX22.1`

ProfDrOetker, 2023-02-16:
Is similar if not identical to #630

IonuDragomir1, 2023-02-16:
Hi, @leeallen1988 !
As you can see, I have encountered the same [issue](https://github.com/frangoteam/FUXA/issues/630).
I have run some tests using S7-1500 and S7-1200 PLCs and I am able to add more than 20 tags. My configuration where I get the error is:
-last version of Fuxa
-Vipa PLC programmed with Step7
What is your configuration? Try to download into your S7-1500 a simple program with a DB where to read more than 20 values from.

unocelli, 2023-02-16:
Hi,
I suppose that the problem is that ReadMultiVars have to split the request, depending on the amount of tags.
Siemens S7 protocol has a limit on the maximum number of variables that can be read in a single request (size limit of the PDU)

leeallen1988, 2023-02-17:
[image]
I solved this problem by adding multiple PLCs with the same IP address，add 20 variables to each container,and it works very well.My program contains M points and DB points, etc.

IonuDragomir1, 2023-02-17:
@unocelli 
I have a S7_1500, S7_1200 and a Vipa PLC.
I made a connection for each of them. For S7_1500 I made around 100 tags and don't get any error. The same happens for S7_1200.
When I connect to Vipa and try to read more than 20 tags, I get the error. So this leads to the conclusion that something from PLS's hardware configuration or program is causing this error, But I couldn't figure it out yet the source.

leeallen1988, 2023-02-17:
> I have a S7_1500, S7_1200 and a Vipa PLC.
> I made a connection for each of them. For S7_1500 I made around 100 tags and don't get any error. The same happens for S7_1200.
> When I connect to Vipa and [...]

Whether the S7 1200PLC has checked the allow put and get options？and does the program contain M points? I will try other configration,like only contains DB points.

IonuDragomir1, 2023-02-17:
@leeallen1988 
`I solved this problem by adding multiple PLCs with the same IP address，add 20 variables to each container,and it works very well.`
This is the solution I found but is not ok. If you'd have an application with 1000 tags, 500 connection should be made. 
`Whether the S7 1200PLC has checked the allow put and get options？` Without this option you couldn't be able to communicate with the PLC.
`does the program contain M points?`
My working examples, no. My faulty example, yes.

unocelli, 2023-02-26:
Hi,
we have pushed a branch s7-vars with the fix.
we would be grateful if someone could test it and provide feedback.

leeallen1988, 2023-02-27:
> Hi, we have pushed a branch s7-vars with the fix. we would be grateful if someone could test it and provide feedback.

Okay,
thanks  a lot for the upgrades,I'm very glad to test it. I hope this time I can solve my problem smoothly.

unocelli, 2023-02-28:
@IonuDragomir1 the cause is as already described: Siemens S7 protocol has a limit on the maximum number of variables that can be read in a single request (size limit of the PDU). To solve it we split the request

unocelli, 2023-03-19:
I am going to close this as resolved, let us know if you have any issues.

closed 2023-03-19 by unocelli