frangoteam/FUXA #695: how to add background image.

nadeda, 2023-02-16:
how to add background image.

unocelli, 2023-02-16:
Hi,
you can add images as background in a view 
[image]

OptifySudarshanPatil, 2023-06-21:
Hi @unocelli ,
it add image into canvas. 
"set image to canvas background" is not visible.

AkashRaneO, 2023-07-18:
Hey!
@unocelli 
I have an image which I want to keep as the background of the view,
if I use an image and fit it to the view canvas and keep it in back it troubles in overlapping.
Is there anyway by which I can add a picture as a background instead of white color.

[image]
i saw this on the website i wanted to do something like this can you please help me