frangoteam/FUXA #704: Already connected to MQTT, but topic cannot be found？

pystudy123, 2023-02-22:
[image]

[image]

[image]

Already connected to MQTT, using emqx, you can see 2 topics, but the topic cannot be found.
Can you directly fill in the subject content, and then connect, what is wrong?

jumpyBah, 2023-02-23:
I have the same problem too.
The mosquitto broker works fine

schmicky, 2023-02-23:
Hi, I am not from the development team but I have this suggestion which worked for me:
press the Search Icon (i see in the screenshot you already pressed it), this puts FUXA in discovery mode,
then use some other tool to generate events on the addresses you want to add,
for example, I used MQTT Explore to change some values in the addresse. When I did that, Fuxa detected the mqtt traffi and the addresse showed up in the menu and i was able to subscribe to them.

Cheers,
George

schmicky, 2023-02-23:
Please test if you can send the value 0 using MQTT, because this is the issue I have. I can send any numeric value except for 0

unocelli, 2023-02-23:
Hi, I suppose that there are the same issue (#435, #242) and we will finally improving it. we will make a subscription available even without the topics browsing function.

jumpyBah, 2023-02-24:
Works !
Thanks schmicky for the trick and thanks unocelli for the improvement

unocelli, 2023-03-19:
I am going to close this as resolved, let us know if you have any issues.

closed 2023-03-19 by unocelli