frangoteam/FUXA #711: Historical data doesn't work properly.

IonuDragomir1, 2023-02-24:
Hi,

I'm trying to display a graphic width historical data. For this I have created a connection to a S7_1500 Siemens PLC having 1 sec polling. Within connection, a tag with a save value interval of 180 seconds (3 minutes) was created.
[image]
[image]
Then the tag was bound to a chart.
[image]
Chart's settings are: 
[image]
The result is the following:
[image]
As you can see in the below image, the graphic is showing tag's changing value after a second (connection's polling time). So my conclusion would be that the tag is not archived at each 3 minutes. I have used Fuxa v1.1.11 - 1064.

ProfDrOetker, 2023-02-24:
Disable "Save value on change"

IonuDragomir1, 2023-02-24:
@ProfDrOetker, thanks for your solution. It works now as expected.

unocelli, 2023-03-19:
I am going to close this as resolved, let us know if you have any issues.

closed 2023-03-19 by unocelli