frangoteam/FUXA #715: Could be used tag's name instead of tag's id?

IonuDragomir1, 2023-02-27:
When I want to call a tag inside a script using a function, I get tag's ID. Did you consider using tag's name instead of its ID? 
[image]
The reason I'm asking this is because in case I reuse the script code in another project, I'd have to add each tag again in script because the IDs don't correspond anymore. I suppose it'd not be the case if tag's name would be used.

unocelli, 2023-02-28:
Hi, 
but then the function parameters should be deviceName and tagName, otherwise you receive an array of tags.

IonuDragomir1, 2023-03-01:
Hi @unocelli,
In all SCADA software, a tag is called by its name. I agree that each tag should have an unique id, but it shouldn't be displayed for user. 
So if an user has added 3 devices (D1, D2, D3). In D1 can be created  several tags (T1, T2, T3). In D2, the user shouldn't be allowed to create tags with names: T1, T2, T3. So if I want to call tag T1 in a script, I should see it by its name, not by its ID. 
Below you can see a screenshot from WinCC where a tag with same name cannot be created and another screenshot with how a tag is called in a script by name.
[image]
[image]
Once I have tried to read some tag's values using an API. It's very hard to do this because the devices and tags were represented by ID, not by name.

unocelli, 2023-03-01:
to uniquely identify a tag you can either use the name or a uuid, both are inefficient. a uuid is also definitely advantageous, example when you want to copy a device automatically. two devices can both have the same Tagname 'Temperature'.

closed 2023-10-11 by unocelli

TomasAdvance, 2023-12-25:
Hi Fuxa team,
will be posible add function getTagName from TagId. Will be helpfull in some cases in scripts.
BR
T

unocelli, 2023-12-30:
Hi, I would be grateful if you could add it as a new issue.