frangoteam/FUXA #718: Server's console gets stuck sometimes

IonuDragomir1, 2023-03-02:
Sometimes console gets stuck . In this moment, having the focus on console, if any key is pressed, it gets back to normal.
The following two photos presents what the console was displaying when got stuck.
[image]

and

[image]
This happens only when I run Fuxa 1.1.12 on Windows 10. I didn't encounter this problem when running on Linux. 
After clicking any key in console shown in first picture, I got this:
[image]

unocelli, 2023-03-07:
Hi,
have you already tried to install the latest version of node-opcua v2.91.1?

unocelli, 2023-10-11:
I’m closing this issue due to inactivity

closed 2023-10-11 by unocelli