frangoteam/FUXA #719: TypeError: Cannot set property 'value' of undefined

gabrielbalinca, 2023-03-03:
Hi,
I'm getting this message in console:

[image]

This error appears only when I assign an external tag's value to a server internal tag. The associations is made with a script.
```
var temp;
temp=$getTag('t_31b9efe4-f4c141b8' /* Local Kep Server - Test1 */);
$setTag('t_bdb44cbd-56eb4622' /* FUXA - int */, temp);
```
When I go to Connections to see internal tag's value but I do get the error in console.  Although I get the error, the script runs correctly and the value is transferred to the internal tag. Why I am getting this error?

unocelli, 2023-03-03:
Hi,
internal tag are available only in frontend, we have to fix it so that you cannot choose in script editor the internal device tags.
I would use fuxa server tags.

gabrielbalinca, 2023-03-03:
For testing I used tags created in FUXA server and I am aware of the fact that I cannot use frontend tags in scripts.
[image]

unocelli, 2023-03-19:
I am going to close this as resolved, let us know if you have any issues.

closed 2023-03-19 by unocelli