frangoteam/FUXA #726: date/time control

patson12, 2023-03-09:
i am going to add date/time control in control expansion panel.. but it shows me blank in editor.. can you help me to add this new control ? or can you add this to in your version??
[image]

unocelli, 2023-03-09:
Great, a datetime control is also in our TODO list, we can try to help you. We have created a 'datetime-control' branch where you can do the Pull Request with your changes.

patson12, 2023-03-11:
can you help me to change current  date time live time in launch project

unocelli, 2023-10-11:
is there any news? can you please explain why you don't solve it with a script?

unocelli, 2023-11-01:
date/time control are in the last version available

closed 2023-12-31 by unocelli