frangoteam/FUXA #727: Can not connect to OPC UA Server

top58525785, 2023-03-10:
HI
I AM USING another NI OPC SERVER SOFTWARE
DOES IT SUPPORT NODE-OPCUA TO CONNECT?

[image]

i meet such problem

2023-03-10T06:44:45.092Z [ERR] 'OPC01' connect failure! TypeError: Cannot read properties of undefined (reading 'position')
2023-03-10T06:44:45.092Z [ERR] 'OPC01' try to connect error! TypeError: Cannot read properties of undefined (reading 'position')

What does it mean?

hxjackcn, 2023-03-12:
You can test it with a common opc-ua client to make sure that the opc-ua server is fine. I haven't found this problem so far.

top58525785, 2023-03-15:
@hxjackcn Thanks for your reply
Do you also using NI OPC SERVER to connect?
When I set the opc configuration
Also it happens error messge as bellow
[image]

top58525785, 2023-09-21:
> You can test it with a common opc-ua client to make sure that the opc-ua server is fine. I haven't found this problem so far.

Which OPC Server do you use?

unocelli, 2023-10-11:
I assume the issue has been resolved in the meanwhile. I’m closing this issue due to inactivity.

closed 2023-10-11 by unocelli