frangoteam/FUXA #732: Chart pop-up legend is missing values

microdeth, 2023-03-14:
Hi, the chart pop-up legend shows the values only if they are integer. If any fractional part presents they are missing from the pop-up legend.
[image]

unocelli, 2023-03-14:
Hi, Thanks for the report. I can reproduce it and I will fix it as soon as possible.

unocelli, 2023-03-18:
Hi,
I can't reproduce any more the issue, in any case now I have removed in chart legend the formatting of digits (the reason for the issue) being configurable in tag property.

microdeth, 2023-03-19:
Hi Umberto,
First, thanks for the great job you've done so far -- I'm looking for the
free SCADA system for my projects and yours is the most promising so far :)
Sorry, if I misbehave in any manner. I'm very new to Github and probably
don't know how things are being done there. You said it's fixed, then I
checked (after update) and saw it's not -- that's why I raised a flag
again. Mb I need to wait for the next release? Please let me know.

Anyway, you're doing the cool things and I'd prefer to stay in contact if
you don't mind :)

Yours,
Dima.

unocelli, 2023-03-19:
Hi, no problem.
can you check if you have the latest version?
[image]

microdeth, 2023-03-20:
Hi Umberto,
I've updated FUXA once more (now minor ver is 12) and everything is great
now!
Thank you :)
BTW, does FUXA stands for `fuchia'?

Regards,
Dima.

unocelli, 2023-03-21:
Hi, **F**rangoteam UX **A**pplication

unocelli, 2023-03-21:
I am going to close this as resolved. let us know if you have any issues.

closed 2023-03-21 by unocelli

microdeth, 2023-03-22:
Thanks, clear.
Dina.