frangoteam/FUXA #733: Node-Red Integration

MatthewReed303, 2023-03-14:
Hi @unocelli 

Have you considered embedding Node-Red into the server? This would allow a lot more flexibility for IIOT data collection, processing data, Databases and logic etc  

If it's not possible or easy how about Fuxa Plugin for Node-Red? If Node-Red is running on the Fuxa Server they can communicate directly and expose all the Tags configured in Fuxa for read and write? This could be done via websocket connection and json? 

I know this can be setup using a OPC connection etc ( which I have tested ) But a Native Integration which it's polling is a lot faster etc would be far superior. 

Thanks for the awesome work on the project so far!

PLCMercenary, 2023-03-18:
@MatthewReed303, I briefly had a similar thought about implementing FUXA from within a Node-Red instance as another UI option, or through the UIBuilder, potentially some-other sort of integration running both in the same node environment. Just curious - is there more you could share with me about what you would want as an MVP from a Node-Fuxa hookup? 

I would be happy to join your efforts and contribute to this idea.

MatthewReed303, 2023-03-18:
Hi @PLCMercenary, I guess either way would work fine or even run separately but exist on same server. The big one is the communication between the two platforms. Having native communication from Node-Red to Fuxa via a tag browser and vice versa, browse Node-Red global variables from within Fuxa would allow access to many other data connections such as different databases, different PLC protocols etc. The user can use flows to integrate logic on the SCADA interface for batching/recipes and schedular etc 

>  is there more you could share with me about what you would want as an MVP from a Node-Fuxa hookup?

1: first thing would be direct communication with all data types from IEC 61131 [https://en.wikipedia.org/wiki/IEC_61131-3](url) most PLC platforms now support this and would make PLC communication from Node-Red to fuxa very easy. 

2: Then having a tag browsing system similar to OPC where you can browse tags in the Fuxa Server and browse tags setup in Node-Fuxa from the Fuxa Server. Possible to import/export tags via CSV for setting up large amounts of tags. 

The problem I found with OPC was the slow response. So something like WebSocket which is bidirectional asynchronous connection with the server would be really fast and already native for web applications etc 

What are your thoughts on this? Unfortunately I'm really new to web stuff and javascript etc so wouldn't know where to start implementing something like this.

PLCMercenary, 2023-03-18:
@MatthewReed303 thanks for the discussion and quick reply! On the weekend no less! I am following your concern involving tag browsing and tag / global variable sharing. 

> 1: first thing would be direct communication with all data types from IEC 61131 [https://en.wikipedia.org/wiki/IEC_61131-3](url) most PLC platforms now support this and would make PLC communication from Node-Red to fuxa very easy.

I am assuming you are probably more experienced on the OT side of the DMZ? Controls Eng, PLC Programmer, SCADA Dev, or the like? (just guessing, its not an insult or anything). I am trying to follow what you mean by "this would make PLC communication from Node-Red to Fuxa very easy"

I think I would approach this project to 1) have deep data integration. 2) have Fuxa run as an editor and application, and thats it. Let nodered handle all of the backend, data and scripting and integrate Fuxa as an editor and industrial frontend

MatthewReed303, 2023-03-18:
@PLCMercenary Yes most of my experience is with PLC and SCADA systems etc I have used Node-Red for all sorts of things, from reading RS232 scales and processing data and passing it through. I come across a lot of different brand PLCs ( mainly pumping stations ) I have been using Node-Red as a bridge to link these older devices. 

I think the main things missing from Fuxa for me to begin using it in the field and sponsor the project more, is the lack of an ODBC Database driver to access different SQL data bases and a scheduler. 

I would suspect moving Fuxa into Node-Red would be quite a big job? How would the frontend editor workout being Angular etc? I think it would be awesome having Fuxa built into NR, but would there be any limitations? Would it just be a plugin into Node-Red. 

What is your background and need for the Integration into Node-Red?