frangoteam/FUXA #736: About "AUTO" property on trend graph Y ranges scaling.

mariorl, 2023-03-15:
The auto property as a fixed option is useful as it always work. This is, whatever datalogging vars values will take, you will never get the trend lines off of the canvas, as it adapts the Y ranges to get traces inside. 
But in certain cases it is not desirable, because many measurements like PH, level etc, may vary very little, for example 0'2, so the graph stretchs the Y axis range in a way that appears as the measurements have got a great value change, but it didn't do.
[image]

unocelli, 2023-03-15:
Hi,
you mean that it would be useful to be able to set the min and max of the y-axis?

mariorl, 2023-03-16:
> Hi, you mean that it would be useful to be able to set the min and max of the y-axis?

Yes

unocelli, 2023-05-14:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-05-14 by unocelli