frangoteam/FUXA #742: No Support for Read-Only Filesystem?

bobf1, 2023-03-18:
Having field device HMI's is a huge benefit, and FUXA could easily achieve this. However I cannot seem to get it to run on a read-only filesystem, which is virtually a requirement of field HMI's since one does not know when the power will be turned off. A read-only filesystem is mostly immune to unscheduled power outages.

Considering the wide application of running FUXA on a read-only filesystem, is this something the developers could easily implement?

Things I've tried, with no success:
1. looking through the various config files for a setting, nothing obvious stands out
2. had linux list all open files in a running Fuxa server instance, then symbolic link all those folders to /var/tmp/fuxa (a tmpfs).

bobf1, 2023-03-18:
[FUXA Log - RW Filesystem.txt](https://github.com/frangoteam/FUXA/files/11009836/FUXA.Log.-.RW.Filesystem.txt)
[FUXA Log - RO Filesystem.txt](https://github.com/frangoteam/FUXA/files/11009838/FUXA.Log.-.RO.Filesystem.txt)

Attached are the logs from FUXA for a RW filesystem that works, then that same filesystem made read-only and it does not work. These logs are on a system that has the following directories all symlinked to /var/tmp/fuxa/:
* _appdata
* _db
* _logs
* _reports

MatthewReed303, 2023-04-01:
Hi @bobf1 did you have any luck getting this to work? I have had a thought of making 2x partitions one for the RO and RW, install the OS on the RO and Run docker with Fuxa on the RW partition

bobf1, 2023-04-02:
Currently playing with automated installation of FUXA to a RAM based filesystem, which obviously is a poor long-term solution.  We're trying to use FUXA as a front-end HMI, which requires the system to gracefully handle sudden power loss.  As such, ANY filesystem that is RW is a no-no.

MatthewReed303, 2023-04-03:
I have had good run with industrial EMMC and using system partition. Running OS in RO and application in the RW, have had quite a few industrial RPI touch screens stand up well so far and still in operation 5 years later. But I agree RO is the way to go and finding a way to achieve this would be great