frangoteam/FUXA #748: Editor issue

top58525785, 2023-03-21:
Version: FUXA-1.1.11-3
HI are u meet with this problem?
Index is OK
but the editor can not come out,always  flashing...
Look at the vedio? How to fix? 

https://user-images.githubusercontent.com/54358179/226567543-7da77dcb-f599-4d5d-aa54-cc12a0384f71.mp4

top58525785, 2023-03-21:
> Hi,
>  Check if the server is running (maybe the console got stuck).

Thanks
The server is ok 

发自我的iPhone

unocelli, 2023-03-21:
Hi, 
the red progressbar on the top show that you have problem to communicate with the server, which version are you using?

top58525785, 2023-03-23:
Version: FUXA-1.1.11-3

top58525785, 2023-03-25:
Also when i using fux 1.1.12 it is not easy to install~~

unocelli, 2023-05-14:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-05-14 by unocelli