frangoteam/FUXA #750: Cannot Browse Bacnet Items on Distech Controls ECY-303

odobrev, 2023-03-21:
Hello there.

I'm trying to evaluate FUXA to see if it could be used as a secondary solution for data visualization (or maybe if it proves to be reliable enough as a primary solution) instead of existing Desigo CC (Siemens) and PSO (Schneider Electrics) Scadas that we are currently using and that are quite annoying to use and really expensive to maintain (I'm not even talking about the cost for each View update ou Device change...).

I successfully validated Modbus TCP on a AnyBus Gateway so I was pretty happy about the speed of the link and the data polling but we do have (for now not so much since the project is still being worked on) about 30-40 ECY-303 from Distech Controls that are only being Bacnet enabled (no Modbus TCP available).

When I'm configuring the connection, I'm able to establish a successful one as shown on this screen copy :
[image]

But when I'm trying to browse tags in server, I get a list with names that don't correspond to real ones :
[image]
[image]

Also, when I'm trying to open a device in list, let's say "Device 502", it keeps showing loader and the server side trips and error :
[image]
[image]

And some time after that, you get an overload warning :
[image]
Also, when you try to open other "Devices" you get this overload error again and again.

It's really similar to other tickets that I've seen on issues history (#652 , #114 , etc.) but the error message seems different this time.

I tried switching to different versions of node (from 12 up to 18) but it did not change anything.
I tried every version of FUXA from (1.0.x to the current version 1.1.12) and it does not seem to solve the problem.

I checked that my Bacnet network was working and devices could be read with INNEA BACNET EXPLORER : 
[image]
I also tried to mess with timeout settings in **runtime/devices/bacnet/index.js** file and some other files where I could find values of timeouts or intervals for polling without success so I'm guessing that it isn't really a "timeout" or "overload" issue but something more complicated than that.

So right now, I'm kinda stuck on my testing because I can't pull data from some devices.

Do you have an idea of what's wrong ?

If you need, I could provide you with more tests / information.
Just tell me what I need to do and I'll try it on this test server.

Greetings.

unocelli, 2023-03-22:
Hi, Thanks for the report.
we currently use a simple simulator for testing and it probably doesn't cover all cases. I can propose you to organise a session and anlize by debugging with you, in any case, I would try to understand why the tag browse does not work (line 113).

odobrev, 2023-03-24:
Hi.

Thanks for the response.
I'm sorry I didn't come back to you sooner but I wanted to check and try some other things before I give you my response.

I tried to see if I could connect from other IOT platforms and I kinda solved my problem by programming node-red as a Bacnet to OPC UA gateway. Then I can just connect from FUXA OPC Connector and build HMI and Alarms from there.
I used it not only for Distech Equipment but also to read Data from Siemens PXC4.E16 PLC.
The only inconvenience from this approach is that it takes more time to setup / configure / maintain and you have to use an external explorer tool to get settings for data reading.

I'm using this flow to read Energy Meters every 24h at a specific time (can't read them too often since they cost about 2 000 € each and they don't have external power; Every query sends a trigger signal that turns them on and they use a little bit of energy from internal battery that can't be changed or charged...so If the battery dies, you get to pay 2000 € + the cost of certification by a regulation entity...) : https://nextcloud.ods-pro.fr/s/ExNaoox4my3Y696
I'm using these palettes : 
- node-red-contrib-bacnet : to read values from equipment
- node-red-contrib-opcua-server : as a connector for FUXA SCADA
- node-red-dashboard : just to create a quick dashboard to confirm that my values are OK

The difference in settings that I can see between FUXA and Node-Red is that you configure a "Device" and a "Client".
In my flow I had to set "Client" to local machine which is actually the default setting and that works perfectly :
[image]
If I set the client configuration as the PLC (IP of PLC, Broadcast of subnet, etc.) I get the same result as on FUXA actually.
The flow throws a payload with the list of Bacnet devices it could find on network by using broadcast...
So the problem might be somewhere around these settings.
In the credits of node-red-contrib-bacnet package, it is stated the following : 
[image]
I did my research and it seems that they are talking about the node-bacstack from [fh1ch](https://github.com/fh1ch/node-bacstack)  driver that can be found on github. If I'm not mistaken, you also use this driver. It might just be that your implementation misses some fine tuning parameters compared to node-red.
You may want to compare your implementation to that of node-red package.

If you want, we can organize a debug session but I'm not that good in spoken English.

Greetings.

unocelli, 2023-03-29:
Hi, sorry for the delay,
Let me check the missing settings and the implementation. 
Language should not be a problem, my French is better than English.

unocelli, 2023-05-14:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-05-14 by unocelli