frangoteam/FUXA #752: Enable/disable snap to grid action

IonuDragomir1, 2023-03-22:
Hello,
When I create an interface, I noticed that snap to grid action is enabled. It'd be useful if this option could be enabled/disabled by user by clicking a button from top toolbar. This feature would be used when very detailed interface requires on creating very much precision.

aniston, 2023-09-12:
Not sure if 'snap to grid' aligns (snaps) an item to the grid or is the background grid there only for indication/placement purposes ? in my Docker Version 1.1.15-1269 it does not 'snap' to the grid although the grid is visible and only the grid visibility can be turned on or off, tried with Firefox, Chrome and Edge.

It would be nice to have items actually snap to the background grid for optimal placement.

aniston, 2026-09-30:
> In the last version is available a step setting in the header, I suppose this is what is meant

yup, exactly, thanks for the addition and sorry for the late response ;(