frangoteam/FUXA #755: Modbus RTU over TCP/IP needed.

RoystonDaCosta, 2023-03-24:
When using serial servers (i.e. Ethernet to RS485 devices) the data need to be sent in the RTU packet format but over TCP/IP.
Is this something that can be done?

unocelli, 2023-03-29:
Hi,
I think it would be better to ask to the people who implemented the driver used in fuxa [modbus-serial](https://github.com/yaacov/node-modbus-serial)

rvbatista, 2023-04-07:
I believe that Modbus-serial driver already implement RTU over TCP.

rvbatista, 2023-04-11:
Implemented in pull request #780
This is my first collaboration with a meaningful opensource project, please be kind.

unocelli, 2023-04-13:
@rvbatista you are welcome :wink:

RoystonDaCosta, 2023-04-23:
Thanks a lot for this feature

rvbatista, 2023-04-23:
@RoystonDaCosta please give a feedback after you test.

RoystonDaCosta, 2023-04-25:
@rvbatista I have tested #780 and it works as expected. Thanks a ton.

rvbatista, 2023-04-25:
Thanks for the feedback.

unocelli, 2023-05-14:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-05-14 by unocelli