frangoteam/FUXA #757: PayloadTooLargeError: request entity too large

IonuDragomir1, 2023-03-24:
Hello,
When I open a project, I get this error:

[image]

It's been already many days since I'm trying to figure out the cause of the problem. The tests and project were run on the last version of FUXA. My project contains several screens with interfaces on them. I have noticed that the error was caused of the screens that contain images. For example, on Screen 1 (S1) I had an image with some other graphical object over it. If the image is deleted, the project is saved and I want to open it, I don't get anymore the error. In the below table you can see  screen's size with and without image. Project's JSON file size is about 32 MB.

[image]

In case you want to test, please create a new project and on a screen add an image of 7MB, for example, and then clone the screen several time. If there is just a screen with a photo, you won't get the error. It will appear if there are 4 or more screens containing the image.
So what shoud be done in order to be able to add images on interface without getting the above error?

unocelli, 2023-03-24:
Hi, Thanks for the report.
Currently the editor embeds the image in the view, the solution is not ideal for the reason you have correctly described. we have planed to give the option to save images on the server and bind it via links.

unocelli, 2023-09-11:
sorry for the delay, actually the size of the image you import can cause different kinds of problems. we currently have a task open to save the file on the server and only associate the link in the view. The disadvantage will be that when exporting the project you will lose the image.

unocelli, 2023-10-11:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-10-11 by unocelli