frangoteam/FUXA #762: How to bind the numeric elements in SVG and JSON one by one

fleffer, 2023-03-27:
I used webAPI to get data,but the the order of data in JSON is not fixed.I want to bind the svg and data by id,how can I implement it?

unocelli, 2023-03-30:
Hi, I don't understand what you mean, you can send the json content

fleffer, 2023-03-31:
For example，there are three valves and a JSON file containing valve information, I seted every 10 sec to get the JSON one time. The JSON was set to change,since the order of the information of each valve in the JSON is not fixed, and because the value of the tag is obtained in array order (e.g. list[0]), the correspondence between the valve and the label will be wrong.
What I want to do is ordering the JSON under my rules when get JSON to make sure the correspondence between the valve and the label is right.
Now,I have changed the value of obtained JSON in "/api/project" by changing the code of 'proApp.get()'.The original value of "status/star_count/collaborator_count" was "1/4/0", I changed its value with id, the value of "status/star_count/collaborator_count" was changed to  ["4/0/1"]
`runtime.project.getProject(req.userId, groups).then(result => {
                let order = {'t_6add0b33-a07143ad':{"tag":"status","gID":"star_count"},
                            't_2e8efc6d-11104423':{"tag":"star_count","gID":"collaborator_count"},
                            't_2fc00efc-67f74262': {"tag":"collaborator_count","gID":"status"}}
                let obj = null;
                obj = JSON.parse(JSON.stringify(result))
                for(let Name in obj.devices["d_a3b3f9ea-f3bd4085"].tags){
                    let currN = order[Name].gID
                    let resID = null
                    for(let IDName in result.devices["d_a3b3f9ea-f3bd4085"].tags){
                        if(result.devices["d_a3b3f9ea-f3bd4085"].tags[IDName].name == currN){
                            resID = IDName
                            break
                        }
                    }
                    let v = result.devices["d_a3b3f9ea-f3bd4085"].tags[resID].value
                    obj.devices["d_a3b3f9ea-f3bd4085"].tags[Name].value  = v
                    //console.log(v,Name,resID)
                }
                result = obj`

[image]
But the tags' values still not change in the device  settting. I want to konw can I change the tags'value or not.If so,what shuld I do or where could I change the code to change the tags'value?
[image]

fleffer, 2023-03-31:
I changed JSON's data by changing code of _readRequest(). It worked well. 
`let obj = null;
                    obj = JSON.parse(JSON.stringify(res.data))
                    let order = {
                        't_702322bf-06d34415': { "currN":"status","gID": "star_count" },
                        't_a945d0ae-a4b74a7d': { "currN":"star_count","gID": "collaborator_count" },
                        't_e39a8299-faf64898': { "currN":"collaborator_count","gID": "status" }
                    }
                    for (let Name in order) {
                        let currN = order[Name].currN
                        let gID = order[Name].gID
                        obj[currN] = res.data[gID]
                    }
                    res.data = obj`

unocelli, 2023-05-14:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-05-14 by unocelli