frangoteam/FUXA #764: Can't add historical values in a table

IonuDragomir1, 2023-03-29:
Hello,

I have added on my interface two tables. One displays some real-time values, and the other one should display historical values. 
As you can see in the below image, I managed to add three tags on Data table by clicking on + button from the left  lower corner of Table customizer window. 

[image]

The + button doesn't exist on History table's Table customizer window. Therefore I can't add any tags to be displayed in History table. 

[image]

Am I missing something or this is a bug?

unocelli, 2023-03-30:
Hi,
for the History table you can only add columns with Tags, the rows are filled with the history values: timestamp, values

IonuDragomir1, 2023-03-31:
Thanks for replying, @unocelli!
Yes, it was my mistake. I managed to add historical values, but I encountered a problem. When the table control is added on interface, columns order is: timestamp and label. If with this column order, I add a tag for label (Tags) column, everything is OK. This is what I get with default columns order:

[image]

The problem appears if switch columns order. This is what I get with switched columns order:

[image]

I changed column order by deleting the Data/Time column and then added it back.

unocelli, 2023-04-12:
Hi, Thanks for the report, about the column order we will fix it as soon as possible.
**Later Edit**: you are able to define the column width

unocelli, 2023-10-11:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-10-11 by unocelli