frangoteam/FUXA #768: improvement_  Opcua performance needs to be improved

hxjackcn, 2023-03-30:
When the performance of the opcua server is fine, after my test, when the number of points exceeds 4000 in FUXA, the speed is very slow.

hxjackcn, 2023-04-05:
When "disable the broadcast of all tags", the client web will not be displayed. What is the function of this function?
[image]

hxjackcn, 2023-04-05:
> Hi, I suppose it also happens if you disable the broadcast of all tags (explained in #772)?

When "disable the broadcast of all tags", the client web will not be displayed. What is the function of this function?
[image]

unocelli, 2023-04-12:
Hi, sadly, the feature still has some problems. Have any values been displayed?

closed 2023-05-04 by hxjackcn