frangoteam/FUXA #777: How to add my own dynamic SVG?

fleffer, 2023-04-07:
I tried to add dynamic SVG to Resource，but it turned to static SVG.If I want to used my own dynamic SVG,how can I do?

unocelli, 2023-04-10:
Hi, what do you mean with dynamic SVG?

fleffer, 2023-04-13:
Sorry for my poor English.I want to add SVG which like the pipe stays moving.The SVG I wanted to add contains the label "<animate>"

fleffer, 2023-04-13:
Sorry for my poor English.I wanted to add SVG which stays moving ,such as the pipe.The SVG I wanted to add contains the lable "animate"

unocelli, 2023-04-16:
The pipe is quite complicated, if you want to change the animation you have to look in PipeComponent where the style is added to the DOM object. Whereas if you want to create your own control you can refer to iframe or Table

fleffer, 2023-04-17:
> The pipe is quite complicated, if you want to change the animation you have to look in PipeComponent where the style is added to the DOM object. Whereas if you want to create your own control you can refer to iframe or Table

It's a great method, it works great, thank you for your reply

fleffer, 2023-04-18:
> The pipe is quite complicated, if you want to change the animation you have to look in PipeComponent where the style is added to the DOM object. Whereas if you want to create your own control you can refer to iframe or Table
The iframe has scroll bars, can I adjust the style of the scroll bars?

unocelli, 2023-04-20:
I think so

closed 2023-05-14 by unocelli