frangoteam/FUXA #778: [BUGS] Button Toggle / SVG Issues etc

MatthewReed303, 2023-04-08:
Hi, There seems to be a bug when using mouse click and toggle value. If you create new project and add just a button and output value and an internal fuxa tag button wont update the value. But if you add toggle switch and use the switch once then the button works fine. 

I have also noticed with some SVG I add the move controls for objects start working in opposite directions ( this happens for all objects ) have to create new project to stop it. Example you move object to the right with mouse the object actually moves to the left

There are a bunch of weird little glitches and things not working as expected. SVG fill colours will not change, but if I test with blank shape the fill works. The hide and show. If I set image to hide and show when value is true, the image does not hide until value is toggled to true and then back to false. ( the initial value is set to 0 ) 

I'm using latest docker version Version 1.1.13-1176

rvbatista, 2023-04-08:
Have the same issue of the toggle became with the value false and not 0 and stop working.

unocelli, 2023-04-08:
Hi, Thanks for the report.
We have removed a feature that caused that some values were not sent to the frontend. Please test again with the new version on the master branch

unocelli, 2023-04-08:
@MatthewReed303 could you please send us an example project with the issues?

MatthewReed303, 2023-04-08:
Hi @unocelli still have all the same issues with pulling latest. Here is a link to some screen recordings:
[https://nextcloud.mrea.co.nz/s/zZHonrNH8SER9yP](url)

unocelli, 2023-04-09:
@MatthewReed303 I have just updated the docker version (latest) with the master branch
[image]

About the editor, try clearing the browser cache

MatthewReed303, 2023-04-09:
@unocelli Thanks button is working correct again and now project also saves the image. I have linked the project which has weird issues with move direction when the SVG is added

[https://nextcloud.mrea.co.nz/s/ejXM4Gt4K26eNWy](https://nextcloud.mrea.co.nz/s/ejXM4Gt4K26eNWy)

unocelli, 2023-04-09:
@MatthewReed303 Thanks for the example, I can reproduce it. 
You have added an svg image that cannot be seen but which has sent the editor in strange status.
[image]

MatthewReed303, 2023-04-09:
Hi @unocelli here is a link to the SVG File. You will also notice it does not display the motor correctly also. 

[https://nextcloud.mrea.co.nz/s/iAWSNiBxxKYXoYs](https://nextcloud.mrea.co.nz/s/iAWSNiBxxKYXoYs)

The fill colour would not work on the motor also. If I add normal SVG shape the fill works correctly. In the mean time I was you going to hide and show all the different states until the fill works. I will have grey/green/orange/red/black so very time consuming to add all these different colour SVGs for different Items. 

For the fill to work is there an common ID in the SVGs you are using on the existing SVG shapes? Can I for example rename the motor body ID to the ID the other shapes use and will the fill work?

MatthewReed303, 2023-04-09:
What if we had an option under shape property's to enter in the SVG ID we wish to edit with fill/blink etc 

[image]

we would also need an SVG ID field under action and blink also

Would this be a suitable option until such time imported SVGs can be fully edited like the one's included in the shapes?

unocelli, 2023-04-10:
@MatthewReed303 About the svg image imported that the move controls is working in opposite directions, I investigated and noticed that the image format is version 1.1 and is not supported in fuxa editor, if you get the latest version of inkscape you can save for version 1.0 and everything works as expected

unocelli, 2023-05-18:
I added #789 issue for clarity.

closed 2023-05-18 by unocelli

MatthewReed303, 2023-06-11:
Hi @unocelli just coming back to the project to continue and trying to save svg as v1.0  but can't find option in the latest inkscape to save as that version. Where did you set the v1.0??