frangoteam/FUXA #779: Input Gauges Information Missing in gaugesManager.getMappedGaugeSignals

studuino, 2023-04-10:
Hi,
gaugesManager.getMappedGaugesSignals (or) hmiService.getMappedVariables doesn't return  'input' gauges information(id, name, source, etc.).

Or is it meant to be that way.
FUXA version 1.1.12
Thanks

unocelli, 2023-04-11:
Hi, the function should return a list of variable (tags) that have a bind to a control or shape of view, to every variable will be add the device name as source attribute. is used to show the variable list in Lab -> Test

studuino, 2023-04-11:
Thank you @unocelli I understand now.
I was hoping to have attributes of text box input as well as tags.
My goal is to write to modbus holding register via text box and button(with click event: sent from input).
But I found out writing to holding register straight away using this.hmiService.putSignalValue

I think because of the auto tag id generation in newer releases I am a little lost. Now I get back on track.
Keep up great work.

closed 2023-04-11 by studuino