frangoteam/FUXA #782: Customized dashboard cannot be retrieved.

rgw123, 2023-04-11:
https://user-images.githubusercontent.com/112950683/231191318-a68103cc-7913-46e4-afdb-38c0e10253cb.mp4

Customized dashboard cannot be retrieved. 

[image]

unocelli, 2023-04-11:
Hi, I do not understand what you mean for dashboard

rgw123, 2023-04-12:
Thank you for your reply，I  send you a video。
dashboard,It means' gauge 'in fuxa，
I added a component in Fuxa, but the newly added component cannot be dragged out. I can't find where my problem lies, and then I saw the prefix. I don't know how or where this prefix is defined

unocelli, 2023-04-12:
The editor insert a svg component with a id and prefix so by initialize the view will be search id and add the component to the item. The prefix is only a simple way to find where to append the component. if you debug you will see the DOM structure of `ele`.

`let htmlBag = Utils.searchTreeStartWith(ele, this.prefixD);`

rgw123, 2023-04-13:
But if I want to define a gauge component myself, how should I define the prefix? Or does the prefix not need to be defined, but if it is not defined, I cannot drag the defined component out. I don't know the rules for defining the prefix, or in which file, such as prefixD: 'D-BAG', why is it D-BAG? What is the basis for it

rgw123, 2023-04-13:
Sorry to bother, is this code just a drag and drop code? But I have customized the filling and still cannot drag it out. What files do I need to pay attention to when writing custom components

unocelli, 2023-04-14:
Hi,
you must follow how one of the latest controls has been implemented: iframe (html-iframe) and Table (html-table).

rgw123, 2023-04-16:
Thank you for your suggestion. I am now able to drag it out, but it cannot be displayed. Can you help me? Thank you for your reply

unocelli, 2023-04-20:
you have to share your code, I can't imagine what the issue is

rgw123, 2023-04-21:
Excuse me
Thank you for your reply. If I want to add some attributes of the instrument pointer on the dashboard, how should I add them, such as the width of the instrument pointer and the color gradient of the instrument pointer.

unocelli, 2023-04-24:
For the color gradient you need a ColorPicker control that support gradient too (I have not found).
Keep in mind that fuxa uses the [gauge.js](https://github.com/bernii/gauge.js) library and if you want to add an attribute that does not exist in the library you have to modify the library too.

rgw123, 2023-04-26:
Thank you for your reply. If I want to define some new features, they must be defined in the gauges. js file, right? Which gradient color did you say ColorPicker was not found? I don't understand what you mean. Is this feature included in this FUXA project?What I mean is that the color of the dashboard pointer can be changed to a gradient color。Thank you very much for answering my question. I hope we can stay in touch

rgw123, 2023-04-26:
Excuse me，I would like to ask where the specific functions in this gague.js are defined, such as staticZones. Is this function directly defined in this file

unocelli, 2023-05-02:
```
Gauge.prototype.render = function() {
 ...
  if (this.options.staticZones) {
```

rgw123, 2023-05-11:
Hello! I would like to confirm whether Fuxa supports WebAssembly and whether there is a wasm.ts file in this project that can be used to call this project through WebAssembly.Thank you for your response.

rgw123, 2023-05-12:
I'm sorry to bother you again. I would like to know where the modes like "select", "fhpath", "line", "circle" in the General list are defined in the code. I'm a student researching FUXA and I think it's great. I'm trying to customize and modify it. Thank you for your help.
The problem I'm currently facing is that my icon (SVG file) can be displayed in the General list but cannot trigger the click function and cannot be dragged out. I don't know where the code for "click="setMode("****")"" is triggered. Can you translate this into English?

unocelli, 2023-05-14:
you must follow how one of the latest controls has been implemented: iframe (html-iframe) and Table (html-table).
(click)="setMode('own_ctrl-iframe')"

rgw123, 2023-05-15:
Sure, thank you for your response. I would like to ask about the click trigger function inside. I have found several files such as ext-bundle-min.js, but these are compressed files. Is there an uncompressed version of the ext-bundle.js file available?

rgw123, 2023-06-06:
How can I add other protocols to this FUXA project?

unocelli, 2023-06-18:
Hi, sorry for the delay...it seems more like a discussion...in server there is a template device

closed 2023-06-18 by unocelli