frangoteam/FUXA #789: Bind Property and Actions to SVG sub components

unocelli, 2023-04-13:
What if we had an option under shape property's to enter in the SVG ID we wish to edit with fill/blink etc 

[image]

we would also need an SVG ID field under action and blink also

Would this be a suitable option until such time imported SVGs can be fully edited like the one's included in the shapes?

_Originally posted by @MatthewReed303 in https://github.com/frangoteam/FUXA/issues/778#issuecomment-1501221284_

MatthewReed303, 2023-04-14:
@unocelli will you be adding this for the next release?

unocelli, 2023-04-20:
@MatthewReed303 we have to look at priorities and resources