frangoteam/FUXA #796: Virtual Keyboard does not work within I-Frame

MatthewReed303, 2023-04-18:
I am trying to use a node-red scheduler which also connects to the PLC via OPC-UA. I am trying to embed the scheduler UI into an Fuxa IFrame to be used from within Fuxa. 

[https://www.npmjs.com/package/node-red-contrib-ui-time-scheduler/v/0.4.4](https://www.npmjs.com/package/node-red-contrib-ui-time-scheduler/v/0.4.4)

It works fine when using from PC browser and keyboard etc but I need it to work from a touchscreen display. The virtual keyboard works for other fuxa devices but does not come up when clicking on something inside of the I-Frame

MatthewReed303, 2023-04-24:
Hi @unocelli have you had any time and luck looking into this issue? Thanks

unocelli, 2023-04-26:
I have just tried using the keyboard in fuxa from an external iframe in touch mode and it works as expected

MatthewReed303, 2023-04-26:
Hi @unocelli can you please push the latest to docker which has the keyboard number orders fixed and I will try it again. Thanks!

unocelli, 2023-04-26:
docker pull frangoteam/fuxa:latest

MatthewReed303, 2023-04-26:
Hi @unocelli nope still does not work for me. I have added a video hopefully it help to understand

[FUXA - Google Chrome 2023-04-27 10-11-13.zip](https://github.com/frangoteam/FUXA/files/11337696/FUXA.-.Google.Chrome.2023-04-27.10-11-13.zip)

unocelli, 2023-05-02:
Hi, I see what you mean, the kayboard is embedded in FUXA and only responds to the input control. for your purpose maybe you need to look for browser plugins, like the windows virtual keyboard.

closed 2023-05-14 by unocelli