frangoteam/FUXA #798: page change of link

tayfunonbasioglu, 2023-04-20:
Is there a method to change the page

http://localhost:1881/editor  or  http://localhost:1881/my_custom_page(this is not work)

thax...

unocelli, 2023-04-20:
Hi,
you can't navigate to the view via url but you can go directly to a view: try http://localhost:1881/view?name=[the View name].
You can only via navigation menu. We will add the feature

hxjackcn, 2023-04-20:
http://localhost:1881/my_custom_page](http://localhost:1881/my_custom_page
still not working

unocelli, 2023-04-21:
It is applied to home url http://localhost:1881/home/my_custom_page

hxjackcn, 2023-04-21:
[image]

unocelli, 2023-04-22:
Hi, Thanks for the report. We have fixed

unocelli, 2023-05-14:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-05-14 by unocelli