frangoteam/FUXA #804: Bool values dont execute as expected

PLCMercenary, 2023-04-26:
Trying to color style an object based on a bool. The tag is bound to the object. I have added the interactivity to set the running and stopped colors. after setting the state values to ``` True [1:1], and False [0:0]``` it just changes to look for a numerical range of low 0 hi 0 and low 1 hi 1. 

Which works. If I go into the lab and send a value of 1 or 0 with the tester, the fill color changes correctly. True / False dont do anything. So i changed my program, and now have the values at 1 and 0, But no matter what I do, my tag binding doesnt send the tag value to the front end. the only way I can work the color is to send a 1 or 0 with the tester.

unocelli, 2023-04-26:
Hi, which versions do you use?

PLCMercenary, 2023-04-28:
> Hi, which versions do you use?

Umberto, I am on your latest docker image.. I just updated and pulled last week

unocelli, 2023-05-02:
Check in 'Connections -> Devices settings' if the value of the Tag updated

PLCMercenary, 2023-05-14:
Discovered I was on a much older version than I expected. my local clone of the repo has been hung up from updating for a while and I didnt pay enough attention when I would git stash and git pull then build the latest dockerfile locally. Now that I am actually fetched up to date and on the main branch. I rebuilt the docker image using your dockerfile and this is working along with a bunch of other things I have been poking at for a few days. We can close this issue!

closed 2023-05-14 by PLCMercenary