frangoteam/FUXA #805: Requirement for multi-popup-dialog/card triggeried by multi-tag-value-change

hyperion-thou, 2023-04-26:
Thanks for your great work first.

I need to popup multi-dialog/card  at the same time beacuse of multi-tag-value acceed threshold( a value a dialog/card).
now mabye two ways:
1. Config alarm action 'Show Popup' for each-tag, BUT now only can show one dialog at the same time ( 'Show Card' maybe the solution BUT we have no this options)
2. Pre-set multi-instance of Shapes like below-image and trigger it's Hide/Shown, BUT we can‘t config 'Click & Close'  Shape event for it.
[image]

Above all, hope my describe is clear. Any advise?

Thanks for your great work again. ^_^

unocelli, 2023-05-02:
Sorry for the delay

1. it is not the option that is missing but the management of the content of the cards that have an UUID to be identified, and if you open several cards of the same type you cannot differentiate. currently is not supported and need a refactoring.
2. I don't have understand

hyperion-thou, 2023-05-05:
Thanks for your reply,

For 2, in order to open serveral cards , i also had try to preset multiple-instance of  retangle-shape which combined-with-a-iframe(to mock a card) , then it can control theirs show/hide when some tag-value-change.

but  **no way to config an 'Click & Close' event for it**  , i think it's also a  new feature. 
[image]

unocelli, 2023-05-09:
Hi, this workaround cannot work, multi-card functionality needs to be implemented.

unocelli, 2023-10-11:
duplicates #744

closed 2023-10-11 by unocelli