frangoteam/FUXA #810: Suggestions about deleting data from databases.

IonuDragomir1, 2023-04-28:
Hi,
I'd like to make some suggestions about deleting data from databases.
1. It's not clear to me what data are deleted from DB or what happens with the DB after 1 year from the option shown below:
 
[image]

What data from SQLite db is deleted after 1 year? The SQLite may contain many databases with lots of tables.

2.  
[image]

Clearing all alarms and history feature is very useful, but are many practical situations where the user wants to keep alarms or history from the last months, or a year. So it'd be useful if the user could choose how much time the alarms and history data are stored in database.

unocelli, 2023-04-28:
Hi, 
1. the DAQ storage for SQlite is separated and saved in _db, chunked in files ~pro day and these files are deleted if they are old.
2. We have to add the feature like DAQ, so that you can define how long the alarm history should be maintained.

unocelli, 2023-05-19:
@IonuDragomir1 You will find the feature in the latest commit. let me know if you have any issues.

unocelli, 2023-07-09:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-07-09 by unocelli