frangoteam/FUXA #811: Suggestions about rendering alarms or history alarms

IonuDragomir1, 2023-04-28:
Hi,
The suggestions I'm about to make are available both for alarms and history alarms.
1. It'd be very useful if would run a progress bar (like the indeterminate progress-bar shown in [this example](https://material.angular.io/components/progress-bar/overview)) until alarms are being  displayed. If there are, for example, 10000 alarms/history alarms, the user doesn't know if there are any alarms to be displayed or if the system crashed. The progress bar would tell the user to wait until all alarms are displayed.
2. In the above scenario, 10000 alarms, there isn't available the option of going to the last or first page.

[image]
So beside the next and previous page option, it'd be great if would exist the options: first and last page. This option would be useful also for tags.

unocelli, 2023-05-04:
We added an indicator for alarm loading and a data range picker for history.
The option jump to last and first page is available but hide. I wonder if it is really required?
[image]

IonuDragomir1, 2023-05-05:
Hi,

Thank you for the update!
"The option jump to last and first page is available but hide." What do you mean by "but hide"? Yes it's really useful. Like I said in my topic, if one would have 10000 alarms and would want to see what are the last 10 alarms would have to click 100 times to get to last 10 alarms if there are 100 alarms displayed on page. So the option jump to last and first page should be available and visible.

unocelli, 2023-05-09:
I mean that we have decided to hide the option. with the latest update you can filter the interval and make sure you don't have 10000 alarms.

IonuDragomir1, 2023-05-10:
I don't see why going to first/last page is an option to be hidden. I'll test the update and come back with a review.

IonuDragomir1, 2023-05-22:
Thanks for update! Please show the option jump to last and first page on alarms. This option should be available also for tags declared on devices settings.

[image]

unocelli, 2023-07-09:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-07-09 by unocelli