frangoteam/FUXA #816: Tag Import partial implementation?

Bobblejot, 2023-05-02:
Hi, might just be me but the only way it seems to import and export tag definitions are either with json or csv, which is fine except that with the json template you can specify scaling of the tag but under csv you cant? also under csv you always lose your connection rack and slot numbers if using an s7 connection.

It also seems that you can only import the complete config. you can't just upload some tags independently. which can cause issues, I tried just uploading one tag but it overwrites all.  OPC UA also seems less stable than standard S7 connections.

Overall I am really enjoying using Fuxa but there are a few things that need a little tweaking or it may just be my inexperience with this software.

unocelli, 2023-05-09:
Hi, Tanks for the report. We are agree with your remarks and we will improve the import/export functionality of the tags as soon as possible

lewsut, 2023-11-15:
I agree  here it's clunky. after importing a test csv where I only added on server tag I lost the ability to edit MQTT tags, reverting back doesn't help the pencil icon for editing has gone.