frangoteam/FUXA #817: Elements Group Copy and Tag edit

Bobblejot, 2023-05-04:
When copying an individual element the configuration for fill stroke etc.. is copied but when copying a group of elements this is lost and you have to go through and re-configure all the elements individually again.  It would be nice if this was the same for individual or grouped elements.

A nice to have would be the ability to edit the tag string in situ so that you don't always have to go through the selection process each time, this would speed up working within Fuxa and make life a little easier.

unocelli, 2023-05-09:
Hi, thanks for the suggestions, we are aware of the issues with the grouping functionality and we will improve as soon as possible.

unocelli, 2023-10-11:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-10-11 by unocelli