frangoteam/FUXA #818: change the scrollbar style of the iframe

fleffer, 2023-05-05:
could I change the scrollbar style of the iframe?
[image]

unocelli, 2023-05-09:
Hi, I think that you can't change it. i try 
```
iframe {
    scrollbar-width: thin;
}
```
but don't work

fleffer, 2023-05-10:
I found that when the link is a video link, the scroll bar is gone.Could I use `overflow:hidden` to hide the style bar?

fleffer, 2023-05-10:
I tried to add `overflow:hidden`  to 'body'.The  scrollbar disappeared.
[image]
[image]
But I just made the changes in Chrome. If I can made this change in the code, where should I add the  `overflow:hidden`?

fleffer, 2023-05-10:
maybe replace `iframe {
    scrollbar-width: thin;
}` with  `iframe  body{
    scrollbar-width: thin;
}`will working,I guess.

unocelli, 2023-05-13:
You can also try it with browser developer tools...don't work

unocelli, 2023-07-09:
I’m closing this issue since there is nothing feasible.

closed 2023-07-09 by unocelli