frangoteam/FUXA #819: Issues about bar graph customization

IonuDragomir1, 2023-05-05:
Hi,

I encountered some issue when trying to change bar graph's theme and layout settings. I'm running the last version of Fuxa. Please check the below video:

https://user-images.githubusercontent.com/58516894/236440047-1b20286c-bf1f-4712-b3bf-7e3ca7744b46.mp4

As you can see, when I switch to dark theme, graph's title, X and Y axis disappear. Dark theme I suppose it's meant to display a black background for bar graph.
If I want to customize the theme, I see that title and legend color are changed accordingly. Grid's color doesn't color yellow. Value (Y) and Category (X) axis don't change the color. The same is available for datalabel.
Another issue is about layout's show grid toggle button. When it's switched off shouldn't hidden the grid? The Show tooltip button works as expected.

unocelli, 2023-05-05:
Thanks for the report, we will fix it as soon as possible.

unocelli, 2023-05-14:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-05-14 by unocelli

Wannapayeot, 2024-07-22:
How do you add data to a chart?

unocelli, 2024-08-01:
@Wannapayeot what is not working?

unocelli, 2024-08-17:
@Wannapayeot I planed to make a video tutorial