frangoteam/FUXA #831: Encountered bugs when resizing screens

IonuDragomir1, 2023-05-22:
Hi,

I encountered 3 very important problems with screen resize:
1. When Zoom property is set to Manually. On runtime if I want to zoom screen A, it works. When I switch to screen B, I'm no longer able to zoom the screen. When I get back to screen A, it can't be zoomed anymore. 
2. When Zoom property is set to Auto-resize. On runtime, this feature works only when I change browser's size from one of its corners or when put the browser in full screen mode. For example, to resize screen A, I have to put the browser in full screen or to drag on of its corners. When I switch to screen B, I have to do the same in order to get the screen resized.
3. When I switch from a screen to a multi-views container, the resize settings are not kept,

IonuDragomir1, 2024-03-29:
Hi,

Do you have any update on this topic? It's a very important bug.

robsori, 2024-06-05:
Hi, I noticed that the change of view type screens still does not adapt according to the device when switching from one screen to another by pressing the menu buttons on the left.

unocelli, 2024-08-23:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-08-23 by unocelli