frangoteam/FUXA #834: npm i error

w8w8w8, 2023-05-26:
when I download the fuxa project
cd server and client 
run npm install , there are some error 

D:\code\FUXA\client>npm -v
6.14.18

D:\code\FUXA\client>node -v
v14.21.3

------------
npm ERR! Maximum call stack size exceeded
npm notice Beginning October 4, 2021, all connections to the npm registry - including for package installation - must use TLS 1.2 or higher. You are currently using plaintext http to connect. Please visit the GitHub blog for more information: https://github.blog/2021-08-23-npm-registry-deprecating-tls-1-0-tls-1-1/

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\HP\AppData\Roaming\npm-cache\_logs\2023-05-26T00_43_56_949Z-debug.log

D:\code\FUXA\client>

-------------------

D:\code\FUXA\server>npm install
npm WARN read-shrinkwrap This version of npm is compatible with lockfileVersion@1, but package-lock.json was generated for lockfileVersion@2. I'll try to do my best with it!

> @serialport/bindings-cpp@10.8.0 install D:\code\FUXA\server\node_modules\@serialport\bindings-cpp
> node-gyp-build

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.3.2 (node_modules\fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.3.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})

npm ERR! code ELIFECYCLE
npm ERR! syscall spawn bash
npm ERR! file bash
npm ERR! path bash
npm ERR! errno -4058
npm ERR! @serialport/bindings-cpp@10.8.0 install: `node-gyp-build`
npm ERR! spawn bash ENOENT
npm ERR!
npm ERR! Failed at the @serialport/bindings-cpp@10.8.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

w8w8w8, 2023-05-27:
I change node  version, then get another error:

D:\code\Fuxa\FUXA-1.1.13\client>node -v
v14.17.4

D:\code\Fuxa\FUXA-1.1.13\client>npm -v
6.14.14

D:\code\Fuxa\FUXA-1.1.13\client>python
Python 3.9.0a1 (tags/v3.9.0a1:fd75708, Nov 19 2019, 18:15:20) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.

when run  npm install ,and get some error :

D:\code\Fuxa\FUXA-1.1.13\client>npm install

> esbuild@0.15.5 postinstall D:\code\Fuxa\FUXA-1.1.13\client\node_modules\@angular-devkit\build-angular\node_modules\esbuild
> node install.js

> es5-ext@0.10.62 postinstall D:\code\Fuxa\FUXA-1.1.13\client\node_modules\es5-ext
>  node -e "try{require('./_postinstall')}catch(e){}" || exit 0

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild-linux-64@0.15.5 (node_modules\@angular-devkit\build-angular\node_modules\esbuild\node_modules\esbuild-linux-64):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for esbuild-linux-64@0.15.5: wanted {"os":"linux","arch":"x64"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild-linux-arm@0.15.5 (node_modules\@angular-devkit\build-angular\node_modules\esbuild\node_modules\esbuild-linux-arm):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for esbuild-linux-arm@0.15.5: wanted {"os":"linux","arch":"arm"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild-freebsd-64@0.15.5 (node_modules\@angular-devkit\build-angular\node_modules\esbuild\node_modules\esbuild-freebsd-64):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for esbuild-freebsd-64@0.15.5: wanted {"os":"freebsd","arch":"x64"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild-linux-32@0.15.5 (node_modules\@angular-devkit\build-angular\node_modules\esbuild\node_modules\esbuild-linux-32):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for esbuild-linux-32@0.15.5: wanted {"os":"linux","arch":"ia32"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild-android-arm64@0.15.5 (node_modules\@angular-devkit\build-angular\node_modules\esbuild\node_modules\esbuild-android-arm64):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for esbuild-android-arm64@0.15.5: wanted {"os":"android","arch":"arm64"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild-freebsd-arm64@0.15.5 (node_modules\@angular-devkit\build-angular\node_modules\esbuild\node_modules\esbuild-freebsd-arm64):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for esbuild-freebsd-arm64@0.15.5: wanted {"os":"freebsd","arch":"arm64"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild-darwin-64@0.15.5 (node_modules\@angular-devkit\build-angular\node_modules\esbuild\node_modules\esbuild-darwin-64):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for esbuild-darwin-64@0.15.5: wanted {"os":"darwin","arch":"x64"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: @esbuild/linux-loong64@0.15.5 (node_modules\@angular-devkit\build-angular\node_modules\esbuild\node_modules\@esbuild\linux-loong64):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for @esbuild/linux-loong64@0.15.5: wanted {"os":"linux","arch":"loong64"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild-darwin-arm64@0.15.5 (node_modules\@angular-devkit\build-angular\node_modules\esbuild\node_modules\esbuild-darwin-arm64):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for esbuild-darwin-arm64@0.15.5: wanted {"os":"darwin","arch":"arm64"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild-linux-riscv64@0.15.5 (node_modules\@angular-devkit\build-angular\node_modules\esbuild\node_modules\esbuild-linux-riscv64):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for esbuild-linux-riscv64@0.15.5: wanted {"os":"linux","arch":"riscv64"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild-netbsd-64@0.15.5 (node_modules\@angular-devkit\build-angular\node_modules\esbuild\node_modules\esbuild-netbsd-64):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for esbuild-netbsd-64@0.15.5: wanted {"os":"netbsd","arch":"x64"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild-linux-mips64le@0.15.5 (node_modules\@angular-devkit\build-angular\node_modules\esbuild\node_modules\esbuild-linux-mips64le):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for esbuild-linux-mips64le@0.15.5: wanted {"os":"linux","arch":"mips64el"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild-linux-arm64@0.15.5 (node_modules\@angular-devkit\build-angular\node_modules\esbuild\node_modules\esbuild-linux-arm64):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for esbuild-linux-arm64@0.15.5: wanted {"os":"linux","arch":"arm64"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild-windows-32@0.15.5 (node_modules\@angular-devkit\build-angular\node_modules\esbuild\node_modules\esbuild-windows-32):
[... cut ...]
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild@0.15.5 (node_modules\@angular-devkit\build-angular\node_modules\esbuild):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: esbuild@0.15.5 postinstall: `node install.js`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: spawn bash ENOENT

npm ERR! Maximum call stack size exceeded

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\HP\AppData\Roaming\npm-cache\_logs\2023-05-27T12_28_56_762Z-debug.log

can you help me? what can I do ?

unocelli, 2023-05-27:
Hi, you are try to install the client to debug? the server installation worked?

w8w8w8, 2023-05-28:
> Hi, you are try to install the client to debug? the server installation worked?

when I cd server dir, and run npm install ,get some error.

D:\code\Fuxa\FUXA-1.1.13\server>node -v
v14.17.4

D:\code\Fuxa\FUXA-1.1.13\server>npm -v
6.14.14

D:\code\Fuxa\FUXA-1.1.13\server>npm install
npm WARN deprecated debug@4.1.1: Debug versions >=3.2.0 <3.2.7 || >=4 <4.3.1 have a low-severity ReDos regression when used in a Node.js environment. It is recommended you upgrade to 3.2.7 or 4.3.1. (https://github.com/visionmedia/debug/issues/797)
npm WARN deprecated @types/socket.io-parser@3.0.0: This is a stub types definition. socket.io-parser provides its own type definitions, so you do not need this installed.
npm WARN deprecated @types/minipass@3.3.5: This is a stub types definition. minipass provides its own type definitions, so you do not need this installed.
npm WARN deprecated @npmcli/move-file@1.1.2: This functionality has been moved to @npmcli/fs
npm WARN deprecated sourcemap-codec@1.4.8: Please use @jridgewell/sourcemap-codec instead

> @serialport/bindings-cpp@10.8.0 install D:\code\Fuxa\FUXA-1.1.13\server\node_modules\@serialport\bindings-cpp
> node-gyp-build

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@~2.3.2 (node_modules\chokidar\node_modules\fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.3.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: @tootallnate/once@1 (node_modules\http-proxy-agent\node_modules\@tootallnate\once):
npm WARN network SKIPPING OPTIONAL DEPENDENCY: request to https://registry.npmjs.org/@tootallnate%2fonce failed, reason: Client network socket disconnected before secure TLS connection was established

npm ERR! code ELIFECYCLE
npm ERR! syscall spawn bash
npm ERR! file bash
npm ERR! path bash
npm ERR! errno -4058
npm ERR! @serialport/bindings-cpp@10.8.0 install: `node-gyp-build`
npm ERR! spawn bash ENOENT
npm ERR!
npm ERR! Failed at the @serialport/bindings-cpp@10.8.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\HP\AppData\Roaming\npm-cache\_logs\2023-05-28T00_12_49_047Z-debug.log

w8w8w8, 2023-05-28:
error maybe by `npm i pdfmake@^0.2.7 `

when run npm i pdfmake@^0.2.7  ，I get error:

npm ERR! Maximum call stack size exceeded

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\HP\AppData\Roaming\npm-cache\_logs\2023-05-28T00_34_03_728Z-debug.log

w8w8w8, 2023-06-04:
[image]

when I change the os, use ubuntu, cd server dir , run "npm install", I get  some error

unocelli, 2023-06-05:
You can try to delete the node_modules directory and the package-lock.json file from server and install 
again

OptifySudarshanPatil, 2023-06-09:
Hi @w8w8w8 
Try [Install Visual C++ Build Environment: [Visual Studio Build Tools]](https://visualstudio.microsoft.com/thank-you-downloading-visual-studio/?sku=BuildTools)

w8w8w8, 2023-06-11:
I Reinstall the system win10  and  run npm i  , it work.
thank all

closed 2023-06-11 by w8w8w8