frangoteam/FUXA #841: reports

patson12, 2023-06-08:
i have many tags added in reports but it gives zero value in report mail.. do you have something how reports work?
PFA
[image]

unocelli, 2023-06-08:
Hi, have you the registration of the tags enabled?
[image]

patson12, 2023-06-10:
ok now its working fine.. but it seems when select many tags then reports get cut .. like suppose i added 20 tags into reports it prints only 15 tags in landscape mode. remaining 5 should come in next pages and like wise. i think some layout  format issue

unocelli, 2023-06-16:
we could put a limit on the table. i prefer that the users are free to decide when there are too many tags. what would you propose?

unocelli, 2023-07-09:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-07-09 by unocelli

yu-1949, 2024-10-25:
> 好的，现在它工作正常..但是，当选择许多标签时，报告似乎会被删减..就像我在报表中添加 20 个标签一样，它在横向模式下只打印 15 个标签。剩下的 5 个应该会出现在下一页，同样明智。我认为一些布局格式问题
May I ask if you have solved the problem that the value of the download report is all 0 in the master branch? The ones I download now are all 0